thecognifly/YAMSPy #1: Wifi access

ItamarBS2411, 2020-02-04:
Hi,
I saw that the device is written as "tty/..." can I write an IP address instead? so I will transmit the command to ESP module that is on the drone (the ESP is just a bridge) or am I missing something in the setup?
Thanks

ItamarBS2411, 2020-02-06:
I tried your library and it does communicate with the FC smoothly - so it looks great.
However I didn't succeed to arm it when pressed 'a' - it was stuck in "Arming..."
Please let me know if you see the issue that I opened? Attached the screenshot of what I see.
[image]

ricardodeazambuja, 2020-02-06:
I just noticed I forgot to add instructions on how to setup the flight controller :)

The image you shared suggests the flight controller is in failsafe mode because it's not receiving data from a receiver. It needs to receive something from a receiver (I think it's 10Hz minimum) or it will automatically enter failsafe mode.

So, YAMSPy works by pretending to be a receiver that speaks MSP. Therefore it's necessary to change the setup for the receiver (e.g. using betaflight-configurator or inav-configurator) to MSP.

BTW, I never tested this library with older versions of Betaflight or INAV and the newest ones (4.0 and 2.3) were not extensively tested either ;)

closed 2020-03-20 by ricardodeazambuja