thecognifly/YAMSPy #2: crashing reading

lrryhss, 2020-05-16:
I running inav 2.3

And crashing running read_adtitude.py it looks like data is being readed.
[MSPy.log](https://github.com/thecognifly/YAMSPy/files/4637640/MSPy.log)

Traceback (most recent call last):
  File "read_altitude.py", line 45, in <module>
    with MSPy(device=serial_port, loglevel='DEBUG', baudrate=115200) as board:
  File "/usr/local/lib/python3.7/dist-packages/yamspy/__init__.py", line 869, in __enter__
    self.is_serial_open = not self.connect(trials=self.ser_trials)
  File "/usr/local/lib/python3.7/dist-packages/yamspy/__init__.py", line 903, in connect
    self.basic_info()
  File "/usr/local/lib/python3.7/dist-packages/yamspy/__init__.py", line 938, in basic_info
    dataHandler = self.receive_msg()
  File "/usr/local/lib/python3.7/dist-packages/yamspy/__init__.py", line 1089, in receive_msg
    data = received_bytes[di]
IndexError: index out of range

ricardodeazambuja, 2020-05-19:
According to your log file it's failing while receiving the reply for the MSPV2_INAV_ANALOG. Does it happen every time or just randomly?

lrryhss, 2020-05-20:
it happen every time.

lrryhss, 2020-05-20:
btw this is for the Outcast MicroHawk We are using the HAKRC MiniF4  STM32F405

lrryhss, 2020-05-20:
Got it working. It may help if I plug in the ESC ;)

closed 2020-05-20 by lrryhss

ricardodeazambuja, 2020-05-20:
I was thinking about this problem and the message is only really checked (crcError) at the end, therefore it was crashing the whole script instead of only marking it with packet_error.
This should fix the problem: f401e24c02007c0dfdbc1c270140705000120277