thecognifly/YAMSPy #6: Omnibus F4 V3 connection with Raspberry pi4 by using MSP

UcanYusuf, 2020-10-08:
Hello, I am trying to make a otonom drone without person control. Rashperry pi will send images to computer and I will process that image in computer. After that computer send command to raspberry pi. And raspberry pi send command to omnibus for motors.  I have a problem about betaflight and MSP. We tried a lot of things but we still have problems. Betaflight supported our controller, we can get data by using usb port. We are trying to use raspberry pi4 and UART6(on flight controller). We couldn't do it "MSP RX input" configuration. Therefore, when using raspberry, we cannot receive the data like when using the usb port.

[image]

[image]

[image]

[image]

[image]

[image]

ricardodeazambuja, 2020-10-09:
Try to enable "MSP" for the UART6 and see if it works. BTW, be careful because the RPI pins are 3.3V while most flight controllers use 5V and that may burn the pins in your RPI.

UcanYusuf, 2020-10-09:
> Try to enable "MSP" for the UART6 and see if it works. BTW, be careful because the RPI pins are 3.3V while most flight controllers use 5V and that may burn the pins in your RPI.

We were able to control the motors from raspberry using the MultiWii protocol. However, now we only rotate the motors by sending a separate pwm value for each motor. How are we going to send right, left or up commands to the controller with msp while flying? It is not a good practice to send individual pwm values ​​for each motor. For example we want to send "take-off" or "go forward" command. How can we do it without send pwm values?

ricardodeazambuja, 2020-10-09:
Ok, nice to know you already can communicate with the FC :+1: 

Now, using Betaflight you don't have specialized cmds for UP/DOWN/LEFT/RIGHT or TAKE-OFF. Betaflight (at least last time I checked) doesn't offer anything for autonomous drones. If you keep using Betaflight you will need to implement the automatic control using the RPI and only send ROLL/PITCH/THROTTLE and YAW. TAKE-OFF will basically be the same as arming the drone and increasing the throttle until it starts flying.

According to what I can understand, you really want an autonomous drone. Therefore I suggest you checking if you can flash iNav in your FC:
https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20Omnibus%20F4.md

After that, I suggest you to read this info [here](https://github.com/thecognifly/YAMSPy#more-details-about-how-yamspy-works) and finally try the [simpleUI](https://github.com/thecognifly/YAMSPy/blob/master/Examples/simpleUI.py).

Let me know how it goes.

ucanAlp, 2020-10-10:
Hi we tried it with inav and betaflight. Both of them has give similar result. 
With INAV first I enabled MSP on cli with "set receiver_type=MSP" also MSP already enabled at ports tab for USB VCP.
While like this in receiver tab there is no enable input bar(colorful bars yaw pitch throttle...). We can take sensor datas In this way. But when we try to run simpleUI.py it is running without any error but when type "a" or arrow keys nothing happening.
[image]
Also we tried this with betaflight(with the old configuration which allow us arm the quad) in betaflight at receiver tab there is colorful input bars and we can adjust limit values unlike INAV. But the result is same. We trying these with usb cable on pc. It can be related?  And I have one more question. If we decided to write our msp based control program (with betaflight) should we use sensor datas? 
I mean while flying with normal receiver and transmitter when we give a stick input for example we pushed the pitch stick and drone is going forward. For do this with MSP should we use gyro,accelerometer,barometer datas? If so flight controller will be just like a sensor board and Raspberry will do almost everything it's wrong ? Thanks again.

ricardodeazambuja, 2020-10-10:
Here is one example of modes configuration using iNav:
[image]

> AETR (Aileron, Elevator, Throttle and Rudder) is the default for YAMSPy and all RC commands will be expected to start with Roll, Pitch, Throttle and Yaw, exactly in this order. Additionally, the auxiliary channels will come just after Yaw following their own numbering (Aux1, Aux2... for Betaflight or CH5, CH6... for iNAV).

from https://github.com/thecognifly/YAMSPy#more-details-about-how-yamspy-works

ucanAlp, 2020-10-11:
It is working. Thank you very much. We will examine this code more carefully to understand better. I hope we can contribute this issue. Thanks for your interest.

ricardodeazambuja, 2020-10-11:
You are welcome! Please, don't forget to share your work on github when you finish. Send me the link and I can add to the main readme.

closed 2020-10-11 by ricardodeazambuja

UcanYusuf, 2020-10-21:
Hello, the example code working on linux but I am getting this error on windows: 

[image]
 
İs there a solution for windows? Thank you.