thecognifly/YAMSPy #7: Example code working on Linux but not working on Wo

UcanYusuf, 2020-10-21:
Hello, the example code working on linux but I am getting this error on windows:

[image]

İs there a solution for windows? Thank you.

ricardodeazambuja, 2020-10-21:
Ok, I pushed a new version that should work with windows. Please, reinstall and test it ;)
```
pip3 install git+git://github.com/ricardodeazambuja/yamspy --upgrade
```

UcanYusuf, 2020-10-21:
> Ok, I pushed a new version that should work with windows. Please, reinstall and test it ;)
> 
> ```
> pip3 install git+git://github.com/ricardodeazambuja/yamspy --upgrade
> ```
> 
> You may need to install a special version of [curses](https://docs.python.org/3/library/curses.html) to use the [simpleUI.py](https://github.com/thecognifly/YAMSPy/blob/0eda0abb673dcf949f3cf7da880e18b7cd817776/Examples/simpleUI.py#L48) on windows:
> https://pypi.org/project/windows-curses/

İt is working now. Thank you very much. You are awesome 👍

closed 2020-10-21 by ricardodeazambuja