thecognifly/YAMSPy #9: Error sending data and receiving imu data

labiybafakh, 2020-12-06:
I have some issues with your project,
The speed of motor didn't change when the value of throttle, pitch, roll, and yaw have changed.
But your project can detect which board is used.
I use ubuntu 18.04.
I attach some pictures about my setting in betaflight, maybe I did some mistakes.
Thank you.

[image]
[image]
[image]

ricardodeazambuja, 2020-12-15:
Can you receive any other data from the flight controller? What does happen when you try the simpleUI.py?

labiybafakh, 2020-12-15:
I just received firmware version, but I can't send any command or read attitude

ricardodeazambuja, 2020-12-15:
Could you please paste here the firmware info you are receiving?

labiybafakh, 2020-12-15:
it's the information of firmware used
[image]

ricardodeazambuja, 2020-12-16:
It's interesting. According to the screenshot, it's communicating with your flight controller because it receives cpuload, battery, etc. Would you mind trying the "example.py" and posting here the results? I would suggest you to copy from the terminal (shift+ctrl+c) and pasting here as `code` (just click on the `< >` icon above) because it will be a long list of messages.

The "example.py" script will try all the messages, so we will be able to check what works.

labiybafakh, 2020-12-16:
it's the result when I ran the example.py

`{'apiVersion': '1.42.0', 'flightControllerIdentifier': 'BTFL', 'flightControllerVersion': '4.1.1', 'version': 0, 'buildInfo': 'Nov 15 2019 12:34:58', 'multiType': 0, 'msp_version': 0, 'capability': 0, 'cycleTime': 129, 'i2cError': 0, 'activeSensors': 33, 'mode': 0, 'profile': 0, 'uid': [5963827, 808472848, 892941363], 'accelerometerTrims': [0, 0], 'name': 'FUS-FPV', 'displayName': 'JOE PILOT', 'numProfiles': 3, 'rateProfile': 0, 'boardType': 2, 'armingDisableCount': 24, 'armingDisableFlags': 4, 'armingDisabled': False, 'runawayTakeoffPreventionDisabled': False, 'boardIdentifier': 'MK41', 'boardVersion': 0, 'commCapabilities': 3, 'targetName': 'MATEKF411', 'boardName': 'MATEKF411', 'manufacturerId': '', 'signature': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'mcuTypeId': 4, 'mspProtocolVersion': 0, 'cpuload': 20, 'flightModeFlags': []}
MSP_API_VERSION >>> requested!
Received >>> MSP_API_VERSION
MSP_FC_VARIANT >>> requested!
Received >>> MSP_FC_VARIANT
MSP_FC_VERSION >>> requested!
Received >>> MSP_FC_VERSION
MSP_BOARD_INFO >>> requested!
Received >>> MSP_BOARD_INFO
MSP_BUILD_INFO >>> requested!
Received >>> MSP_BUILD_INFO
MSP_NAME >>> requested!
Received >>> MSP_NAME
MSP_BATTERY_CONFIG >>> requested!
Received >>> MSP_BATTERY_CONFIG
MSP_MODE_RANGES >>> requested!
Received >>> MSP_MODE_RANGES
MSP_FEATURE_CONFIG >>> requested!
Received >>> MSP_FEATURE_CONFIG
MSP_BOARD_ALIGNMENT_CONFIG >>> requested!
Received >>> MSP_BOARD_ALIGNMENT_CONFIG
MSP_CURRENT_METER_CONFIG >>> requested!
Received >>> MSP_CURRENT_METER_CONFIG
MSP_MIXER_CONFIG >>> requested!
Received >>> MSP_MIXER_CONFIG
MSP_RX_CONFIG >>> requested!
Received >>> MSP_RX_CONFIG
MSP_LED_COLORS >>> requested!
Received >>> MSP_LED_COLORS
MSP_LED_STRIP_CONFIG >>> requested!
Received >>> MSP_LED_STRIP_CONFIG
MSP_RSSI_CONFIG >>> requested!
Received >>> MSP_RSSI_CONFIG
MSP_ADJUSTMENT_RANGES >>> requested!
Received >>> MSP_ADJUSTMENT_RANGES
MSP_CF_SERIAL_CONFIG >>> requested!
Received >>> MSP_CF_SERIAL_CONFIG
MSP_VOLTAGE_METER_CONFIG >>> requested!
Received >>> MSP_VOLTAGE_METER_CONFIG
MSP_SONAR >>> requested!
Received >>> MSP_SONAR
MSP_PID_CONTROLLER >>> requested!
Received >>> MSP_PID_CONTROLLER
MSP_ARMING_CONFIG >>> requested!
Received >>> MSP_ARMING_CONFIG
MSP_RX_MAP >>> requested!
Received >>> MSP_RX_MAP
MSP_BF_CONFIG >>> requested!
Received >>> ERROR!
MSP_BF_BUILD_INFO >>> requested!
Received >>> ERROR!
MSP_DATAFLASH_SUMMARY >>> requested!
Received >>> MSP_DATAFLASH_SUMMARY
MSP_LOOP_TIME >>> requested!
Received >>> ERROR!
MSP_FAILSAFE_CONFIG >>> requested!
Received >>> MSP_FAILSAFE_CONFIG
MSP_RXFAIL_CONFIG >>> requested!
Received >>> MSP_RXFAIL_CONFIG
MSP_SDCARD_SUMMARY >>> requested!
Received >>> MSP_SDCARD_SUMMARY
MSP_BLACKBOX_CONFIG >>> requested!
Received >>> MSP_BLACKBOX_CONFIG
MSP_TRANSPONDER_CONFIG >>> requested!
Received >>> MSP_TRANSPONDER_CONFIG
Traceback (most recent call last):
  File "example.py", line 75, in <module>
    board.process_recv_data(dataHandler)
  File "/usr/local/lib/python3.6/dist-packages/yamspy/__init__.py", line 1563, in process_recv_data
    processor(self,data) # use it..
  File "/usr/local/lib/python3.6/dist-packages/yamspy/__init__.py", line 2488, in process_MSP_TRANSPONDER_CONFIG
    self.TRANSPONDER['provider'] = self.readbytes(data, size=8, unsigned=True)
  File "/usr/local/lib/python3.6/dist-packages/yamspy/__init__.py", line 1310, in readbytes
    buffer.append(data.pop(0))
IndexError: pop from empty bytearray
`

ricardodeazambuja, 2020-12-16:
I noticed the code haven't been tested on Betaflight for some time, so I tried to fix the errors that you posted here. Could you please pull my latest commit and run the example.py again? Sadly I don't have a flight controller right now to test it.

labiybafakh, 2020-12-17:
I have tested of your new code

`{'apiVersion': '1.42.0', 'flightControllerIdentifier': 'BTFL', 'flightControllerVersion': '4.1.1', 'version': 0, 'buildInfo': 'Nov 15 2019 12:34:58', 'multiType': 0, 'msp_version': 0, 'capability': 0, 'cycleTime': 132, 'i2cError': 0, 'activeSensors': 33, 'mode': 0, 'profile': 0, 'uid': [5963827, 808472848, 892941363], 'accelerometerTrims': [0, 0], 'name': 'FUS-FPV', 'displayName': 'JOE PILOT', 'numProfiles': 3, 'rateProfile': 0, 'boardType': 2, 'armingDisableCount': 24, 'armingDisableFlags': 4, 'armingDisabled': False, 'runawayTakeoffPreventionDisabled': False, 'boardIdentifier': 'MK41', 'boardVersion': 0, 'commCapabilities': 3, 'targetName': 'MATEKF411', 'boardName': 'MATEKF411', 'manufacturerId': '', 'signature': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'mcuTypeId': 4, 'mspProtocolVersion': 0, 'cpuload': 20, 'flightModeFlags': []}
MSP_API_VERSION >>> requested!
Received >>> MSP_API_VERSION
MSP_FC_VARIANT >>> requested!
Received >>> MSP_FC_VARIANT
MSP_FC_VERSION >>> requested!
Received >>> MSP_FC_VERSION
MSP_BOARD_INFO >>> requested!
Received >>> MSP_BOARD_INFO
MSP_BUILD_INFO >>> requested!
Received >>> MSP_BUILD_INFO
MSP_NAME >>> requested!
Received >>> MSP_NAME
MSP_BATTERY_CONFIG >>> requested!
Received >>> MSP_BATTERY_CONFIG
MSP_MODE_RANGES >>> requested!
Received >>> MSP_MODE_RANGES
MSP_FEATURE_CONFIG >>> requested!
Received >>> MSP_FEATURE_CONFIG
MSP_BOARD_ALIGNMENT_CONFIG >>> requested!
Received >>> MSP_BOARD_ALIGNMENT_CONFIG
MSP_CURRENT_METER_CONFIG >>> requested!
Received >>> MSP_CURRENT_METER_CONFIG
MSP_MIXER_CONFIG >>> requested!
Received >>> MSP_MIXER_CONFIG
MSP_RX_CONFIG >>> requested!
Received >>> MSP_RX_CONFIG
MSP_LED_COLORS >>> requested!
Received >>> MSP_LED_COLORS
MSP_LED_STRIP_CONFIG >>> requested!
Received >>> MSP_LED_STRIP_CONFIG
MSP_RSSI_CONFIG >>> requested!
Received >>> MSP_RSSI_CONFIG
MSP_ADJUSTMENT_RANGES >>> requested!
Received >>> MSP_ADJUSTMENT_RANGES
MSP_CF_SERIAL_CONFIG >>> requested!
Received >>> MSP_CF_SERIAL_CONFIG
MSP_VOLTAGE_METER_CONFIG >>> requested!
Received >>> MSP_VOLTAGE_METER_CONFIG
MSP_SONAR >>> requested!
Received >>> MSP_SONAR
MSP_PID_CONTROLLER >>> requested!
Received >>> MSP_PID_CONTROLLER
MSP_ARMING_CONFIG >>> requested!
Received >>> MSP_ARMING_CONFIG
MSP_RX_MAP >>> requested!
Received >>> MSP_RX_MAP
MSP_BF_CONFIG >>> requested!
Received >>> ERROR!
MSP_BF_BUILD_INFO >>> requested!
Received >>> ERROR!
MSP_DATAFLASH_SUMMARY >>> requested!
Received >>> MSP_DATAFLASH_SUMMARY
MSP_LOOP_TIME >>> requested!
Received >>> ERROR!
MSP_FAILSAFE_CONFIG >>> requested!
Received >>> MSP_FAILSAFE_CONFIG
MSP_RXFAIL_CONFIG >>> requested!
Received >>> MSP_RXFAIL_CONFIG
MSP_SDCARD_SUMMARY >>> requested!
Received >>> MSP_SDCARD_SUMMARY
MSP_BLACKBOX_CONFIG >>> requested!
Received >>> MSP_BLACKBOX_CONFIG
MSP_TRANSPONDER_CONFIG >>> requested!
Received >>> MSP_TRANSPONDER_CONFIG
Traceback (most recent call last):
  File "example.py", line 75, in <module>
    board.process_recv_data(dataHandler)
  File "/usr/local/lib/python3.6/dist-packages/yamspy/__init__.py", line 1563, in process_recv_data
    processor(self,data) # use it..
  File "/usr/local/lib/python3.6/dist-packages/yamspy/__init__.py", line 2488, in process_MSP_TRANSPONDER_CONFIG
    self.TRANSPONDER['provider'] = self.readbytes(data, size=8, unsigned=True)
  File "/usr/local/lib/python3.6/dist-packages/yamspy/__init__.py", line 1310, in readbytes
    buffer.append(data.pop(0))
IndexError: pop from empty bytearray
`

ricardodeazambuja, 2020-12-17:
Are you sure you pulled the latest commit? I'm asking that because the [latest commit wouldn't test MSP_BF_CONFIG](https://github.com/thecognifly/YAMSPy/blob/47f24caebf7f5d6b03354f34114448f5cc29c41b/yamspy/__init__.py#L112).

labiybafakh, 2020-12-18:
I'm sorry I copied the wrong result on my PC because I pulled it into my rapsberrypi zero

`{'apiVersion': '1.42.0', 'flightControllerIdentifier': 'BTFL', 'flightControllerVersion': '4.1.1', 'version': 0, 'buildInfo': 'Nov 15 2019 12:34:58', 'multiType': 0, 'msp_version': 0, 'capability': 0, 'cycleTime': 134, 'i2cError': 0, 'activeSensors': 33, 'mode': 0, 'profile': 0, 'uid': [5963827, 808472848, 892941363], 'accelerometerTrims': [0, 0], 'name': 'FUS-FPV', 'displayName': 'JOE PILOT', 'numProfiles': 3, 'rateProfile': 0, 'boardType': 2, 'armingDisableCount': 24, 'armingDisableFlags': 4, 'armingDisabled': False, 'runawayTakeoffPreventionDisabled': False, 'boardIdentifier': 'MK41', 'boardVersion': 0, 'commCapabilities': 3, 'targetName': 'MATEKF411', 'boardName': 'MATEKF411', 'manufacturerId': '', 'signature': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'mcuTypeId': 4, 'mspProtocolVersion': 0, 'cpuload': 20, 'flightModeFlags': []}
MSP_API_VERSION >>> requested!
Received >>> MSP_API_VERSION
MSP_FC_VARIANT >>> requested!
Received >>> MSP_FC_VARIANT
MSP_FC_VERSION >>> requested!
Received >>> MSP_FC_VERSION
MSP_BOARD_INFO >>> requested!
Received >>> MSP_BOARD_INFO
MSP_BUILD_INFO >>> requested!
Received >>> MSP_BUILD_INFO
MSP_NAME >>> requested!
Received >>> MSP_NAME
MSP_BATTERY_CONFIG >>> requested!
Received >>> MSP_BATTERY_CONFIG
MSP_MODE_RANGES >>> requested!
Received >>> MSP_MODE_RANGES
MSP_FEATURE_CONFIG >>> requested!
Received >>> MSP_FEATURE_CONFIG
MSP_BOARD_ALIGNMENT_CONFIG >>> requested!
Received >>> MSP_BOARD_ALIGNMENT_CONFIG
MSP_CURRENT_METER_CONFIG >>> requested!
Received >>> MSP_CURRENT_METER_CONFIG
MSP_MIXER_CONFIG >>> requested!
Received >>> MSP_MIXER_CONFIG
MSP_RX_CONFIG >>> requested!
Received >>> MSP_RX_CONFIG
MSP_LED_COLORS >>> requested!
Received >>> MSP_LED_COLORS
MSP_LED_STRIP_CONFIG >>> requested!
Received >>> MSP_LED_STRIP_CONFIG
MSP_RSSI_CONFIG >>> requested!
Received >>> MSP_RSSI_CONFIG
MSP_ADJUSTMENT_RANGES >>> requested!
Received >>> MSP_ADJUSTMENT_RANGES
MSP_CF_SERIAL_CONFIG >>> requested!
Received >>> MSP_CF_SERIAL_CONFIG
MSP_VOLTAGE_METER_CONFIG >>> requested!
Received >>> MSP_VOLTAGE_METER_CONFIG
MSP_SONAR >>> requested!
Received >>> MSP_SONAR
MSP_PID_CONTROLLER >>> requested!
Received >>> MSP_PID_CONTROLLER
MSP_ARMING_CONFIG >>> requested!
Received >>> MSP_ARMING_CONFIG
MSP_RX_MAP >>> requested!
Received >>> MSP_RX_MAP
MSP_BF_CONFIG >>> requested!
Received >>> ERROR!
MSP_BF_BUILD_INFO >>> requested!
Received >>> ERROR!
MSP_DATAFLASH_SUMMARY >>> requested!
Received >>> MSP_DATAFLASH_SUMMARY
MSP_LOOP_TIME >>> requested!
Received >>> ERROR!
MSP_FAILSAFE_CONFIG >>> requested!
Received >>> MSP_FAILSAFE_CONFIG
MSP_RXFAIL_CONFIG >>> requested!
Received >>> MSP_RXFAIL_CONFIG
MSP_SDCARD_SUMMARY >>> requested!
Received >>> MSP_SDCARD_SUMMARY
MSP_BLACKBOX_CONFIG >>> requested!
Received >>> MSP_BLACKBOX_CONFIG
MSP_TRANSPONDER_CONFIG >>> requested!
Received >>> MSP_TRANSPONDER_CONFIG
Traceback (most recent call last):
  File "example.py", line 75, in <module>
    board.process_recv_data(dataHandler)
  File "/usr/local/lib/python3.7/dist-packages/yamspy/__init__.py", line 1563, in process_recv_data
    processor(self,data) # use it..
  File "/usr/local/lib/python3.7/dist-packages/yamspy/__init__.py", line 2488, in process_MSP_TRANSPONDER_CONFIG
    self.TRANSPONDER['provider'] = self.readbytes(data, size=8, unsigned=True)
  File "/usr/local/lib/python3.7/dist-packages/yamspy/__init__.py", line 1310, in readbytes
    buffer.append(data.pop(0))
IndexError: pop from empty bytearray
`

ricardodeazambuja, 2020-12-18:
I still can see at least the "MSP_BF_CONFIG" in your output, so that means it is not the latest commit because [that particular MSP code was commented out](https://github.com/thecognifly/YAMSPy/blob/47f24caebf7f5d6b03354f34114448f5cc29c41b/yamspy/__init__.py#L112) and there's no other way for example.py script to request that anymore.

labiybafakh, 2020-12-18:
I checked the init file and it already changed,
I tried to delete the folder and clone again.
Maybe it's like what u mean
`{'apiVersion': '1.42.0', 'flightControllerIdentifier': 'BTFL', 'flightController                                                                                       Version': '4.1.1', 'version': 0, 'buildInfo': 'Nov 15 2019 12:34:58', 'multiType                                                                                       ': 0, 'msp_version': 0, 'capability': 0, 'cycleTime': 132, 'i2cError': 0, 'activ                                                                                       eSensors': 33, 'mode': 0, 'profile': 0, 'uid': [5963827, 808472848, 892941363],                                                                                        'accelerometerTrims': [0, 0], 'name': 'FUS-FPV', 'displayName': 'JOE PILOT', 'nu                                                                                       mProfiles': 3, 'rateProfile': 0, 'boardType': 2, 'armingDisableCount': 24, 'armi                                                                                       ngDisableFlags': 4, 'armingDisabled': False, 'runawayTakeoffPreventionDisabled':                                                                                        False, 'boardIdentifier': 'MK41', 'boardVersion': 0, 'commCapabilities': 3, 'ta                                                                                       rgetName': 'MATEKF411', 'boardName': 'MATEKF411', 'manufacturerId': '', 'signatu                                                                                       re': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,                                                                                        0, 0, 0, 0, 0, 0, 0], 'mcuTypeId': 4, 'mspProtocolVersion': 0, 'cpuload': 17, '                                                                                       flightModeFlags': []}
MSP_API_VERSION >>> requested!
Received >>> MSP_API_VERSION
MSP_FC_VARIANT >>> requested!
Received >>> MSP_FC_VARIANT
MSP_FC_VERSION >>> requested!
Received >>> MSP_FC_VERSION
MSP_BOARD_INFO >>> requested!
Received >>> MSP_BOARD_INFO
MSP_BUILD_INFO >>> requested!
Received >>> MSP_BUILD_INFO
MSP_NAME >>> requested!
Received >>> MSP_NAME
MSP_BATTERY_CONFIG >>> requested!
Received >>> MSP_BATTERY_CONFIG
MSP_MODE_RANGES >>> requested!
Received >>> MSP_MODE_RANGES
MSP_FEATURE_CONFIG >>> requested!
Received >>> MSP_FEATURE_CONFIG
MSP_BOARD_ALIGNMENT_CONFIG >>> requested!
Received >>> MSP_BOARD_ALIGNMENT_CONFIG
MSP_CURRENT_METER_CONFIG >>> requested!
Received >>> MSP_CURRENT_METER_CONFIG
MSP_MIXER_CONFIG >>> requested!
Received >>> MSP_MIXER_CONFIG
MSP_RX_CONFIG >>> requested!
Received >>> MSP_RX_CONFIG
MSP_LED_COLORS >>> requested!
Received >>> MSP_LED_COLORS
MSP_LED_STRIP_CONFIG >>> requested!
Received >>> MSP_LED_STRIP_CONFIG
MSP_RSSI_CONFIG >>> requested!
Received >>> MSP_RSSI_CONFIG
MSP_ADJUSTMENT_RANGES >>> requested!
Received >>> MSP_ADJUSTMENT_RANGES
MSP_CF_SERIAL_CONFIG >>> requested!
Received >>> MSP_CF_SERIAL_CONFIG
MSP_VOLTAGE_METER_CONFIG >>> requested!
Received >>> MSP_VOLTAGE_METER_CONFIG
MSP_SONAR >>> requested!
Received >>> MSP_SONAR
MSP_PID_CONTROLLER >>> requested!
Received >>> MSP_PID_CONTROLLER
MSP_ARMING_CONFIG >>> requested!
Received >>> MSP_ARMING_CONFIG
MSP_RX_MAP >>> requested!
Received >>> MSP_RX_MAP
MSP_DATAFLASH_SUMMARY >>> requested!
Received >>> MSP_DATAFLASH_SUMMARY
MSP_LOOP_TIME >>> requested!
Received >>> ERROR!
MSP_FAILSAFE_CONFIG >>> requested!
Received >>> MSP_FAILSAFE_CONFIG
MSP_RXFAIL_CONFIG >>> requested!
Received >>> MSP_RXFAIL_CONFIG
MSP_SDCARD_SUMMARY >>> requested!
Received >>> MSP_SDCARD_SUMMARY
MSP_BLACKBOX_CONFIG >>> requested!
Received >>> MSP_BLACKBOX_CONFIG
MSP_TRANSPONDER_CONFIG >>> requested!
Received >>> MSP_TRANSPONDER_CONFIG
MSP_OSD_CONFIG >>> requested!
Received >>> MSP_OSD_CONFIG
MSP_OSD_CHAR_READ >>> requested!
Received >>> ERROR!
MSP_OSD_CHAR_WRITE >>> requested!
Received >>> MSP_OSD_CHAR_WRITE
[... 98 lines cut ...]
MSP_DEBUG >>> requested!
Received >>> MSP_DEBUG
MSP2_COMMON_MOTOR_MIXER >>> requested!
Received >>> MSP2_COMMON_MOTOR_MIXER
MSP2_COMMON_PG_LIST >>> requested!
Received >>> ERROR!
MSP2_CF_SERIAL_CONFIG >>> requested!
Received >>> ERROR!
MSPV2_INAV_STATUS >>> requested!
Received >>> ERROR!
MSPV2_INAV_OPTICAL_FLOW >>> requested!
Received >>> MSPV2_INAV_OPTICAL_FLOW
MSPV2_INAV_ANALOG >>> requested!
Received >>> MSPV2_INAV_ANALOG
Traceback (most recent call last):
  File "example.py", line 75, in <module>
    board.process_recv_data(dataHandler)
  File "/usr/local/lib/python3.7/dist-packages/yamspy/__init__.py", line 1563, i                                                                                       n process_recv_data
    processor(self,data) # use it..
  File "/usr/local/lib/python3.7/dist-packages/yamspy/__init__.py", line 1698, i                                                                                       n process_MSPV2_INAV_ANALOG
    self.ANALOG['amperage'] = self.readbytes(data, size=16, unsigned=True) / 100                                                                                        # A
  File "/usr/local/lib/python3.7/dist-packages/yamspy/__init__.py", line 1310, i                                                                                       n readbytes
    buffer.append(data.pop(0))
IndexError: pop from empty bytearray
`

labiybafakh, 2020-12-19:
I used this code to read attitude and it's work, but I can't send the data to control motor
[pyMultiWii](https://github.com/nik012003/pyMultiWii)

ricardodeazambuja, 2020-12-19:
Thank you very much for the feedback. I pushed [a new update](https://github.com/thecognifly/YAMSPy/commit/14ab374a2b93bee36c97db2f467c76e0ff3bc807), could you please give it a try by running the example.py script and posting here the results?

labiybafakh, 2020-12-20:
I have tried it, and for the information this flighcontroller using betaflight

`{'apiVersion': '1.42.0', 'flightControllerIdentifier': 'BTFL', 'flightControllerVersion': '4.1.1', 'version': 0, 'buildInfo': 'Nov 15 2019 12:34:58', 'multiType': 0, 'msp_version': 0, 'capability': 0, 'cycleTime': 127, 'i2cError': 0, 'activeSensors': 33, 'mode': 0, 'profile': 0, 'uid': [5963827, 808472848, 892941363], 'accelerometerTrims': [0, 0], 'name': 'FUS-FPV', 'displayName': 'JOE PILOT', 'numProfiles': 3, 'rateProfile': 0, 'boardType': 2, 'armingDisableCount': 24, 'armingDisableFlags': 4, 'armingDisabled': False, 'runawayTakeoffPreventionDisabled': False, 'boardIdentifier': 'MK41', 'boardVersion': 0, 'commCapabilities': 3, 'targetName': 'MATEKF411', 'boardName': 'MATEKF411', 'manufacturerId': '', 'signature': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'mcuTypeId': 4, 'mspProtocolVersion': 0, 'cpuload': 17, 'flightModeFlags': []}
MSP_API_VERSION >>> requested!
Received >>> MSP_API_VERSION
MSP_FC_VARIANT >>> requested!
Received >>> MSP_FC_VARIANT
MSP_FC_VERSION >>> requested!
Received >>> MSP_FC_VERSION
MSP_BOARD_INFO >>> requested!
Received >>> MSP_BOARD_INFO
MSP_BUILD_INFO >>> requested!
Received >>> MSP_BUILD_INFO
MSP_NAME >>> requested!
Received >>> MSP_NAME
MSP_BATTERY_CONFIG >>> requested!
Received >>> MSP_BATTERY_CONFIG
MSP_MODE_RANGES >>> requested!
Received >>> MSP_MODE_RANGES
MSP_FEATURE_CONFIG >>> requested!
Received >>> MSP_FEATURE_CONFIG
MSP_BOARD_ALIGNMENT_CONFIG >>> requested!
Received >>> MSP_BOARD_ALIGNMENT_CONFIG
MSP_CURRENT_METER_CONFIG >>> requested!
Received >>> MSP_CURRENT_METER_CONFIG
MSP_MIXER_CONFIG >>> requested!
Received >>> MSP_MIXER_CONFIG
MSP_RX_CONFIG >>> requested!
Received >>> MSP_RX_CONFIG
MSP_LED_COLORS >>> requested!
Received >>> MSP_LED_COLORS
MSP_LED_STRIP_CONFIG >>> requested!
Received >>> MSP_LED_STRIP_CONFIG
MSP_RSSI_CONFIG >>> requested!
Received >>> MSP_RSSI_CONFIG
MSP_ADJUSTMENT_RANGES >>> requested!
Received >>> MSP_ADJUSTMENT_RANGES
MSP_CF_SERIAL_CONFIG >>> requested!
Received >>> MSP_CF_SERIAL_CONFIG
MSP_VOLTAGE_METER_CONFIG >>> requested!
Received >>> MSP_VOLTAGE_METER_CONFIG
MSP_SONAR >>> requested!
Received >>> MSP_SONAR
MSP_PID_CONTROLLER >>> requested!
Received >>> MSP_PID_CONTROLLER
MSP_ARMING_CONFIG >>> requested!
Received >>> MSP_ARMING_CONFIG
MSP_RX_MAP >>> requested!
Received >>> MSP_RX_MAP
MSP_BF_CONFIG >>> requested!
Received >>> ERROR!
MSP_BF_BUILD_INFO >>> requested!
Received >>> ERROR!
MSP_DATAFLASH_SUMMARY >>> requested!
Received >>> MSP_DATAFLASH_SUMMARY
MSP_LOOP_TIME >>> requested!
Received >>> ERROR!
MSP_FAILSAFE_CONFIG >>> requested!
Received >>> MSP_FAILSAFE_CONFIG
MSP_RXFAIL_CONFIG >>> requested!
Received >>> MSP_RXFAIL_CONFIG
MSP_SDCARD_SUMMARY >>> requested!
Received >>> MSP_SDCARD_SUMMARY
MSP_BLACKBOX_CONFIG >>> requested!
Received >>> MSP_BLACKBOX_CONFIG
MSP_TRANSPONDER_CONFIG >>> requested!
Received >>> MSP_TRANSPONDER_CONFIG
Traceback (most recent call last):
  File "example.py", line 75, in <module>
    board.process_recv_data(dataHandler)
  File "/usr/local/lib/python3.6/dist-packages/yamspy/__init__.py", line 1563, in process_recv_data
    processor(self,data) # use it..
  File "/usr/local/lib/python3.6/dist-packages/yamspy/__init__.py", line 2488, in process_MSP_TRANSPONDER_CONFIG
    self.TRANSPONDER['provider'] = self.readbytes(data, size=8, unsigned=True)
  File "/usr/local/lib/python3.6/dist-packages/yamspy/__init__.py", line 1310, in readbytes
    buffer.append(data.pop(0))
IndexError: pop from empty bytearray
`

ricardodeazambuja, 2020-12-20:
Ok, I added exception catching to that part of the code. Now, I would expect the script will just ignore problematic messages instead of crashing. Before you test example.py with this new commit, I suggest you to delete the "MSPy.log" file and post here the contents of the new file generated to help me debug the problem. Thanks!

labiybafakh, 2020-12-20:
Thanks for always trying to help me, it's the result of the code

`[DEBUG] [2020-12-20 20:42:12,276]: RAW message sent: bytearray(b'$M<\x00\x01\x01')
[DEBUG] [2020-12-20 20:42:12,284]: State: 0 - byte received (at 1608464532.284725): 36
[DEBUG] [2020-12-20 20:42:12,284]: State: 1 - byte received (at 1608464532.284725): 77
[DEBUG] [2020-12-20 20:42:12,284]: State: 2 - byte received (at 1608464532.284725): 62
[DEBUG] [2020-12-20 20:42:12,284]: State: 3 - byte received (at 1608464532.284725): 3
[DEBUG] [2020-12-20 20:42:12,284]: State: 4 - byte received (at 1608464532.284725): 1
[DEBUG] [2020-12-20 20:42:12,284]: State: 7 - byte received (at 1608464532.284725): 0
[DEBUG] [2020-12-20 20:42:12,284]: State: 8 - byte received (at 1608464532.284725): 1
[DEBUG] [2020-12-20 20:42:12,284]: State: 8 - byte received (at 1608464532.284725): 42
[DEBUG] [2020-12-20 20:42:12,285]: State: 9 - byte received (at 1608464532.284725): 41
[DEBUG] [2020-12-20 20:42:12,285]: Message received (length 3) - Code 1
[DEBUG] [2020-12-20 20:42:12,285]: RAW message sent: bytearray(b'$M<\x00\x02\x02')
[DEBUG] [2020-12-20 20:42:12,300]: State: 0 - byte received (at 1608464532.3007598): 36
[DEBUG] [2020-12-20 20:42:12,300]: State: 1 - byte received (at 1608464532.3007598): 77
[DEBUG] [2020-12-20 20:42:12,300]: State: 2 - byte received (at 1608464532.3007598): 62
[DEBUG] [2020-12-20 20:42:12,300]: State: 3 - byte received (at 1608464532.3007598): 4
[DEBUG] [2020-12-20 20:42:12,300]: State: 4 - byte received (at 1608464532.3007598): 2
[DEBUG] [2020-12-20 20:42:12,300]: State: 7 - byte received (at 1608464532.3007598): 66
[DEBUG] [2020-12-20 20:42:12,301]: State: 8 - byte received (at 1608464532.3007598): 84
[DEBUG] [2020-12-20 20:42:12,301]: State: 8 - byte received (at 1608464532.3007598): 70
[DEBUG] [2020-12-20 20:42:12,301]: State: 8 - byte received (at 1608464532.3007598): 76
[DEBUG] [2020-12-20 20:42:12,301]: State: 9 - byte received (at 1608464532.3007598): 26
[DEBUG] [2020-12-20 20:42:12,301]: Message received (length 4) - Code 2
[DEBUG] [2020-12-20 20:42:12,301]: RAW message sent: bytearray(b'$M<\x00\x03\x03')
[DEBUG] [2020-12-20 20:42:12,316]: State: 0 - byte received (at 1608464532.316726): 36
[DEBUG] [2020-12-20 20:42:12,316]: State: 1 - byte received (at 1608464532.316726): 77
[DEBUG] [2020-12-20 20:42:12,316]: State: 2 - byte received (at 1608464532.316726): 62
[DEBUG] [2020-12-20 20:42:12,316]: State: 3 - byte received (at 1608464532.316726): 3
[DEBUG] [2020-12-20 20:42:12,316]: State: 4 - byte received (at 1608464532.316726): 3
[DEBUG] [2020-12-20 20:42:12,316]: State: 7 - byte received (at 1608464532.316726): 4
[DEBUG] [2020-12-20 20:42:12,316]: State: 8 - byte received (at 1608464532.316726): 1
[DEBUG] [2020-12-20 20:42:12,316]: State: 8 - byte received (at 1608464532.316726): 1
[DEBUG] [2020-12-20 20:42:12,317]: State: 9 - byte received (at 1608464532.316726): 4
[DEBUG] [2020-12-20 20:42:12,317]: Message received (length 3) - Code 3
[DEBUG] [2020-12-20 20:42:12,317]: RAW message sent: bytearray(b'$M<\x00\x05\x05')
[DEBUG] [2020-12-20 20:42:12,332]: State: 0 - byte received (at 1608464532.3327336): 36
[DEBUG] [2020-12-20 20:42:12,332]: State: 1 - byte received (at 1608464532.3327336): 77
[DEBUG] [2020-12-20 20:42:12,332]: State: 2 - byte received (at 1608464532.3327336): 62
[DEBUG] [2020-12-20 20:42:12,332]: State: 3 - byte received (at 1608464532.3327336): 26
[DEBUG] [2020-12-20 20:42:12,332]: State: 4 - byte received (at 1608464532.3327336): 5
[DEBUG] [2020-12-20 20:42:12,332]: State: 7 - byte received (at 1608464532.3327336): 78
[DEBUG] [2020-12-20 20:42:12,332]: State: 8 - byte received (at 1608464532.3327336): 111
[DEBUG] [2020-12-20 20:42:12,332]: State: 8 - byte received (at 1608464532.3327336): 118
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 32
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 49
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 53
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 32
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 50
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 48
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 49
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 57
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 49
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 50
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 58
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 51
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 52
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 58
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 53
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 56
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 49
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 101
[DEBUG] [2020-12-20 20:42:12,333]: State: 8 - byte received (at 1608464532.3327336): 53
[... cut ...]
[DEBUG] [2020-12-20 20:42:12,940]: State: 2 - byte received (at 1608464532.9407659): 62
[DEBUG] [2020-12-20 20:42:12,941]: State: 3 - byte received (at 1608464532.9407659): 1
[DEBUG] [2020-12-20 20:42:12,941]: State: 4 - byte received (at 1608464532.9407659): 82
[DEBUG] [2020-12-20 20:42:12,941]: State: 7 - byte received (at 1608464532.9407659): 0
[DEBUG] [2020-12-20 20:42:12,941]: State: 9 - byte received (at 1608464532.9407659): 83
[DEBUG] [2020-12-20 20:42:12,941]: Message received (length 1) - Code 82`

ricardodeazambuja, 2020-12-20:
So, is `example.py` running until the end now?

labiybafakh, 2020-12-20:
yes, I think it's like last commit until transponder config
`{'apiVersion': '1.42.0', 'flightControllerIdentifier': 'BTFL', 'flightControllerVersion': '4.1.1', 'version': 0, 'buildInfo': 'Nov 15 2019 12:34:58', 'multiType': 0, 'msp_version': 0, 'capability': 0, 'cycleTime': 133, 'i2cError': 0, 'activeSensors': 33, 'mode': 0, 'profile': 0, 'uid': [5963827, 808472848, 892941363], 'accelerometerTrims': [0, 0], 'name': 'FUS-FPV', 'displayName': 'JOE PILOT', 'numProfiles': 3, 'rateProfile': 0, 'boardType': 2, 'armingDisableCount': 24, 'armingDisableFlags': 4, 'armingDisabled': False, 'runawayTakeoffPreventionDisabled': False, 'boardIdentifier': 'MK41', 'boardVersion': 0, 'commCapabilities': 3, 'targetName': 'MATEKF411', 'boardName': 'MATEKF411', 'manufacturerId': '', 'signature': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'mcuTypeId': 4, 'mspProtocolVersion': 0, 'cpuload': 17, 'flightModeFlags': []}
MSP_API_VERSION >>> requested!
Received >>> MSP_API_VERSION
MSP_FC_VARIANT >>> requested!
Received >>> MSP_FC_VARIANT
MSP_FC_VERSION >>> requested!
Received >>> MSP_FC_VERSION
MSP_BOARD_INFO >>> requested!
Received >>> MSP_BOARD_INFO
MSP_BUILD_INFO >>> requested!
Received >>> MSP_BUILD_INFO
MSP_NAME >>> requested!
Received >>> MSP_NAME
MSP_BATTERY_CONFIG >>> requested!
Received >>> MSP_BATTERY_CONFIG
MSP_MODE_RANGES >>> requested!
Received >>> MSP_MODE_RANGES
MSP_FEATURE_CONFIG >>> requested!
Received >>> MSP_FEATURE_CONFIG
MSP_BOARD_ALIGNMENT_CONFIG >>> requested!
Received >>> MSP_BOARD_ALIGNMENT_CONFIG
MSP_CURRENT_METER_CONFIG >>> requested!
Received >>> MSP_CURRENT_METER_CONFIG
MSP_MIXER_CONFIG >>> requested!
Received >>> MSP_MIXER_CONFIG
MSP_RX_CONFIG >>> requested!
Received >>> MSP_RX_CONFIG
MSP_LED_COLORS >>> requested!
Received >>> MSP_LED_COLORS
MSP_LED_STRIP_CONFIG >>> requested!
Received >>> MSP_LED_STRIP_CONFIG
MSP_RSSI_CONFIG >>> requested!
Received >>> MSP_RSSI_CONFIG
MSP_ADJUSTMENT_RANGES >>> requested!
Received >>> MSP_ADJUSTMENT_RANGES
MSP_CF_SERIAL_CONFIG >>> requested!
Received >>> MSP_CF_SERIAL_CONFIG
MSP_VOLTAGE_METER_CONFIG >>> requested!
Received >>> MSP_VOLTAGE_METER_CONFIG
MSP_SONAR >>> requested!
Received >>> MSP_SONAR
MSP_PID_CONTROLLER >>> requested!
Received >>> MSP_PID_CONTROLLER
MSP_ARMING_CONFIG >>> requested!
Received >>> MSP_ARMING_CONFIG
MSP_RX_MAP >>> requested!
Received >>> MSP_RX_MAP
MSP_BF_CONFIG >>> requested!
Received >>> ERROR!
MSP_BF_BUILD_INFO >>> requested!
Received >>> ERROR!
MSP_DATAFLASH_SUMMARY >>> requested!
Received >>> MSP_DATAFLASH_SUMMARY
MSP_LOOP_TIME >>> requested!
Received >>> ERROR!
MSP_FAILSAFE_CONFIG >>> requested!
Received >>> MSP_FAILSAFE_CONFIG
MSP_RXFAIL_CONFIG >>> requested!
Received >>> MSP_RXFAIL_CONFIG
MSP_SDCARD_SUMMARY >>> requested!
Received >>> MSP_SDCARD_SUMMARY
MSP_BLACKBOX_CONFIG >>> requested!
Received >>> MSP_BLACKBOX_CONFIG
MSP_TRANSPONDER_CONFIG >>> requested!
Received >>> MSP_TRANSPONDER_CONFIG
Traceback (most recent call last):
  File "example.py", line 75, in <module>
    board.process_recv_data(dataHandler)
  File "/usr/local/lib/python3.6/dist-packages/yamspy/__init__.py", line 1563, in process_recv_data
    processor(self,data) # use it..
  File "/usr/local/lib/python3.6/dist-packages/yamspy/__init__.py", line 2488, in process_MSP_TRANSPONDER_CONFIG
    self.TRANSPONDER['provider'] = self.readbytes(data, size=8, unsigned=True)
  File "/usr/local/lib/python3.6/dist-packages/yamspy/__init__.py", line 1310, in readbytes
    buffer.append(data.pop(0))
IndexError: pop from empty bytearray
`

ricardodeazambuja, 2020-12-20:
Are you sure you pulled the latest commit?

labiybafakh, 2020-12-20:
Oh I'm sorry Mr. Ricardo, I didn't know if u did the lastest commit.
Maybe if u want to use teamviewer and u're not busy, I'm free

`{'apiVersion': '1.42.0', 'flightControllerIdentifier': 'BTFL', 'flightControllerVersion': '4.1.1', 'version': 0, 'buildInfo': 'Nov 15 2019 12:34:58', 'multiType': 0, 'msp_version': 0, 'capability': 0, 'cycleTime': 126, 'i2cError': 0, 'activeSensors': 33, 'mode': 0, 'profile': 0, 'uid': [5963827, 808472848, 892941363], 'accelerometerTrims': [0, 0], 'name': 'FUS-FPV', 'displayName': 'JOE PILOT', 'numProfiles': 3, 'rateProfile': 0, 'boardType': 2, 'armingDisableCount': 24, 'armingDisableFlags': 4, 'armingDisabled': False, 'runawayTakeoffPreventionDisabled': False, 'boardIdentifier': 'MK41', 'boardVersion': 0, 'commCapabilities': 3, 'targetName': 'MATEKF411', 'boardName': 'MATEKF411', 'manufacturerId': '', 'signature': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'mcuTypeId': 4, 'mspProtocolVersion': 0, 'cpuload': 19, 'flightModeFlags': []}
MSP_API_VERSION >>> requested!
Received >>> MSP_API_VERSION
MSP_FC_VARIANT >>> requested!
Received >>> MSP_FC_VARIANT
MSP_FC_VERSION >>> requested!
Received >>> MSP_FC_VERSION
MSP_BOARD_INFO >>> requested!
Received >>> MSP_BOARD_INFO
MSP_BUILD_INFO >>> requested!
Received >>> MSP_BUILD_INFO
MSP_NAME >>> requested!
Received >>> MSP_NAME
MSP_BATTERY_CONFIG >>> requested!
Received >>> MSP_BATTERY_CONFIG
MSP_MODE_RANGES >>> requested!
Received >>> MSP_MODE_RANGES
MSP_FEATURE_CONFIG >>> requested!
Received >>> MSP_FEATURE_CONFIG
MSP_BOARD_ALIGNMENT_CONFIG >>> requested!
Received >>> MSP_BOARD_ALIGNMENT_CONFIG
MSP_CURRENT_METER_CONFIG >>> requested!
Received >>> MSP_CURRENT_METER_CONFIG
MSP_MIXER_CONFIG >>> requested!
Received >>> MSP_MIXER_CONFIG
MSP_RX_CONFIG >>> requested!
Received >>> MSP_RX_CONFIG
MSP_LED_COLORS >>> requested!
Received >>> MSP_LED_COLORS
MSP_LED_STRIP_CONFIG >>> requested!
Received >>> MSP_LED_STRIP_CONFIG
MSP_RSSI_CONFIG >>> requested!
Received >>> MSP_RSSI_CONFIG
MSP_ADJUSTMENT_RANGES >>> requested!
Received >>> MSP_ADJUSTMENT_RANGES
MSP_CF_SERIAL_CONFIG >>> requested!
Received >>> MSP_CF_SERIAL_CONFIG
MSP_VOLTAGE_METER_CONFIG >>> requested!
Received >>> MSP_VOLTAGE_METER_CONFIG
MSP_SONAR >>> requested!
Received >>> MSP_SONAR
MSP_PID_CONTROLLER >>> requested!
Received >>> MSP_PID_CONTROLLER
MSP_ARMING_CONFIG >>> requested!
Received >>> MSP_ARMING_CONFIG
MSP_RX_MAP >>> requested!
Received >>> MSP_RX_MAP
MSP_BF_CONFIG >>> requested!
Received >>> ERROR!
MSP_BF_BUILD_INFO >>> requested!
Received >>> ERROR!
MSP_DATAFLASH_SUMMARY >>> requested!
Received >>> MSP_DATAFLASH_SUMMARY
MSP_LOOP_TIME >>> requested!
Received >>> ERROR!
MSP_FAILSAFE_CONFIG >>> requested!
Received >>> MSP_FAILSAFE_CONFIG
MSP_RXFAIL_CONFIG >>> requested!
Received >>> MSP_RXFAIL_CONFIG
MSP_SDCARD_SUMMARY >>> requested!
Received >>> MSP_SDCARD_SUMMARY
MSP_BLACKBOX_CONFIG >>> requested!
Received >>> MSP_BLACKBOX_CONFIG
MSP_TRANSPONDER_CONFIG >>> requested!
Received >>> MSP_TRANSPONDER_CONFIG
Traceback (most recent call last):
  File "example.py", line 75, in <module>
    board.process_recv_data(dataHandler)
  File "/usr/local/lib/python3.6/dist-packages/yamspy/__init__.py", line 1563, in process_recv_data
    processor(self,data) # use it..
  File "/usr/local/lib/python3.6/dist-packages/yamspy/__init__.py", line 2488, in process_MSP_TRANSPONDER_CONFIG
    self.TRANSPONDER['provider'] = self.readbytes(data, size=8, unsigned=True)
  File "/usr/local/lib/python3.6/dist-packages/yamspy/__init__.py", line 1310, in readbytes
    buffer.append(data.pop(0))
IndexError: pop from empty bytearray
`

ricardodeazambuja, 2020-12-20:
You just need to upgrade the package:
`$ sudo pip3 install git+git://github.com/ricardodeazambuja/yamspy --upgrade`
(I think it will also work without the `--upgrade`)
and verify if `example.py` will run until the end.

labiybafakh, 2020-12-20:
oh I'm sorry, now it works
`{'apiVersion': '1.42.0', 'flightControllerIdentifier': 'BTFL', 'flightControllerVersion': '4.1.1', 'version': 0, 'buildInfo': 'Nov 15 2019 12:34:58', 'multiType': 0, 'msp_version': 0, 'capability': 0, 'cycleTime': 126, 'i2cError': 0, 'activeSensors': 33, 'mode': 0, 'profile': 0, 'uid': [5963827, 808472848, 892941363], 'accelerometerTrims': [0, 0], 'name': 'FUS-FPV', 'displayName': 'JOE PILOT', 'numProfiles': 3, 'rateProfile': 0, 'boardType': 2, 'armingDisableCount': 24, 'armingDisableFlags': 4, 'armingDisabled': False, 'runawayTakeoffPreventionDisabled': False, 'boardIdentifier': 'MK41', 'boardVersion': 0, 'commCapabilities': 3, 'targetName': 'MATEKF411', 'boardName': 'MATEKF411', 'manufacturerId': '', 'signature': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'mcuTypeId': 4, 'mspProtocolVersion': 0, 'cpuload': 17, 'flightModeFlags': []}
MSP_API_VERSION >>> requested!
Received >>> MSP_API_VERSION
MSP_FC_VARIANT >>> requested!
Received >>> MSP_FC_VARIANT
MSP_FC_VERSION >>> requested!
Received >>> MSP_FC_VERSION
MSP_BOARD_INFO >>> requested!
Received >>> MSP_BOARD_INFO
MSP_BUILD_INFO >>> requested!
Received >>> MSP_BUILD_INFO
MSP_NAME >>> requested!
Received >>> MSP_NAME
MSP_BATTERY_CONFIG >>> requested!
Received >>> MSP_BATTERY_CONFIG
MSP_MODE_RANGES >>> requested!
Received >>> MSP_MODE_RANGES
MSP_FEATURE_CONFIG >>> requested!
Received >>> MSP_FEATURE_CONFIG
MSP_BOARD_ALIGNMENT_CONFIG >>> requested!
Received >>> MSP_BOARD_ALIGNMENT_CONFIG
MSP_CURRENT_METER_CONFIG >>> requested!
Received >>> MSP_CURRENT_METER_CONFIG
MSP_MIXER_CONFIG >>> requested!
Received >>> MSP_MIXER_CONFIG
MSP_RX_CONFIG >>> requested!
Received >>> MSP_RX_CONFIG
MSP_LED_COLORS >>> requested!
Received >>> MSP_LED_COLORS
MSP_LED_STRIP_CONFIG >>> requested!
Received >>> MSP_LED_STRIP_CONFIG
MSP_RSSI_CONFIG >>> requested!
Received >>> MSP_RSSI_CONFIG
MSP_ADJUSTMENT_RANGES >>> requested!
Received >>> MSP_ADJUSTMENT_RANGES
MSP_CF_SERIAL_CONFIG >>> requested!
Received >>> MSP_CF_SERIAL_CONFIG
MSP_VOLTAGE_METER_CONFIG >>> requested!
Received >>> MSP_VOLTAGE_METER_CONFIG
MSP_SONAR >>> requested!
Received >>> MSP_SONAR
MSP_PID_CONTROLLER >>> requested!
Received >>> MSP_PID_CONTROLLER
MSP_ARMING_CONFIG >>> requested!
Received >>> MSP_ARMING_CONFIG
MSP_RX_MAP >>> requested!
Received >>> MSP_RX_MAP
MSP_DATAFLASH_SUMMARY >>> requested!
Received >>> MSP_DATAFLASH_SUMMARY
MSP_LOOP_TIME >>> requested!
Received >>> ERROR!
MSP_FAILSAFE_CONFIG >>> requested!
Received >>> MSP_FAILSAFE_CONFIG
MSP_RXFAIL_CONFIG >>> requested!
Received >>> MSP_RXFAIL_CONFIG
MSP_SDCARD_SUMMARY >>> requested!
Received >>> MSP_SDCARD_SUMMARY
MSP_BLACKBOX_CONFIG >>> requested!
Received >>> MSP_BLACKBOX_CONFIG
MSP_TRANSPONDER_CONFIG >>> requested!
Received >>> MSP_TRANSPONDER_CONFIG
MSP_OSD_CONFIG >>> requested!
Received >>> MSP_OSD_CONFIG
MSP_OSD_CHAR_READ >>> requested!
Received >>> ERROR!
MSP_OSD_CHAR_WRITE >>> requested!
Received >>> MSP_OSD_CHAR_WRITE
MSP_VTX_CONFIG >>> requested!
Received >>> MSP_VTX_CONFIG
[... 128 lines cut ...]
MSP2_INAV_MC_BRAKING >>> requested!
Received >>> MSP2_INAV_MC_BRAKING
MSP2_INAV_SELECT_BATTERY_PROFILE >>> requested!
Received >>> ERROR!
MSP2_INAV_DEBUG >>> requested!
Received >>> ERROR!
MSP2_BLACKBOX_CONFIG >>> requested!
Received >>> ERROR!
MSP2_INAV_TEMP_SENSOR_CONFIG >>> requested!
Received >>> ERROR!
MSP2_INAV_TEMPERATURES >>> requested!
Received >>> ERROR!
MSP2_INAV_SERVO_MIXER >>> requested!
Received >>> MSP2_INAV_SERVO_MIXER
MSP2_INAV_LOGIC_CONDITIONS >>> requested!
Received >>> MSP2_INAV_LOGIC_CONDITIONS
MSP2_INAV_LOGIC_CONDITIONS_STATUS >>> requested!
Received >>> MSP2_INAV_LOGIC_CONDITIONS_STATUS
MSP2_PID >>> requested!
Received >>> MSP2_PID
MSP2_INAV_OPFLOW_CALIBRATION >>> requested!
Received >>> MSP2_INAV_OPFLOW_CALIBRATION
armingDisableFlags: ['RXLOSS']
Board is rebooting
`

ricardodeazambuja, 2020-12-20:
Thanks for the feedback! I'm glad it worked :+1: 

Now, please, have a look at [this discussion](https://github.com/thecognifly/YAMSPy/issues/10) and try the simpleUI.

labiybafakh, 2020-12-20:
I tried it and succes, but I want to develop independent control, so I'll send the data for each motor and read attitude of the robot. Because I'll implement / develop control algorithm. Can u give me some advice? thank you

ricardodeazambuja, 2020-12-20:
I'm not 100%, but I think you can only read altitude from INAV. 
My suggestion is for you to first try to understand the script `simpleUI.py`.

closed 2021-10-06 by ricardodeazambuja