thecognifly/YAMSPy #10: inav communication

johncook1212, 2020-12-15:
Hi Ricardo,
Hope all is good, I had to take a step our for a while... but now I am back and will update regarding gpiozero etc.
I wanted to ask a quick question - I've changed the drone's firmware form betaflight to INAV. 
The problem that the drone is not arming when I press the associated key.
I think that the problem is that "arming" is associated with "Aux1" while in INAV configurator arming is associated with "ch5". The drone can be armed via RC (Taranis X9 lite).
I've tried to change "aux1" to "ch5" or "CH5". 

What do you think?
John

johncook1212, 2020-12-15:
[image]
Attached the snapshot form INAV configurator and the script...

ricardodeazambuja, 2020-12-16:
INAV seems to be saying your FC is in FAILSAFE mode according to the screenshot from simpleUI. Try to investigate why it would be entering FAILSAFE mode.
The simpleUI.py sends commands (AETR...) in this order `CMDS_ORDER = ['roll', 'pitch', 'throttle', 'yaw', 'aux1', 'aux2']`, therefore you did it correctly by using CH5 and CH6 (the names inside simpleUI don't really matter, just the order it sends at the end). When you asks it to send the arm command it sends `CMDS['aux1'] = 1800` and 1000 for disarm (yours starts at 1800). The flight mode it selects like this:
```
elif char == ord('m') or char == ord('M'):
    if CMDS['aux2'] <= 1300:
        cursor_msg = 'Horizon Mode...'
        CMDS['aux2'] = 1500
    elif 1700 > CMDS['aux2'] > 1300:
        cursor_msg = 'Flip Mode...'
        CMDS['aux2'] = 2000
    elif CMDS['aux2'] >= 1700:
        cursor_msg = 'Angle Mode...'
        CMDS['aux2'] = 1000
``` 
Maybe this is the reason for the FAILSAFE, but it could be something else. When everything is running normally, simpleUI should not show FAILSAFE for the flight mode.

johncook1212, 2020-12-20:
Thank you Ricardo, it was the commands order (AETR) mismatch.
Best,
John

closed 2020-12-20 by johncook1212