thecognifly/YAMSPy #12: MSP over UART (Pins)

Fr3d1sk, 2021-02-22:
Hello,

Has anybody successfully connected to a FC with a raspberry pi over the onboard Pins? I have deactivated bluetooth and changed many settings but still it doesn't connect successfull, it always times out. If I try another UART where I know it should not work it instantly throws an error...

Via USB everything works great, but usb is not safe enough to fly...

ricardodeazambuja, 2021-02-24:
Hi,

I tested both UARTs available in the RPI Zero W without problems. However, you need to be careful in relation to external interference (use short cables and don't pass them close to anything that may generate EMI). In addition to that, I don't know which FC you are using, but RPI uses 3.3V, so you may need [a level shifter](https://www.raspberrypi-spy.co.uk/2018/09/using-a-level-shifter-with-the-raspberry-pi-gpio) if the FC is 5V.

Let me know how if you managed to make it work.

Cheers,
Ricardo

Fr3d1sk, 2021-03-02:
Hello,
Yes the 3.3V were the problem, thank you.

ricardodeazambuja, 2021-03-02:
Enjoy your flight :)

closed 2021-03-02 by ricardodeazambuja