thecognifly/YAMSPy #14: AssertionError: There's something wrong with the serial connection... check the logfile!

miguelwon, 2021-09-08:
Hi,
Can't run any of the examples. For example, [read_altitude.py](https://github.com/thecognifly/YAMSPy/blob/master/Examples/read_altitude.py) example gives the error:

`AssertionError: There's something wrong with the serial connection... check the logfile!`

But the serial connection is working. The log file:

```
[DEBUG] [2021-09-08 10:56:05,496]: RAW message sent: bytearray(b'$M<\x00\x01\x01')
[DEBUG] [2021-09-08 10:56:05,499]: State: 0 - byte received (at 1631094965.4989734): 36
[DEBUG] [2021-09-08 10:56:05,499]: State: 1 - byte received (at 1631094965.4989734): 77
[DEBUG] [2021-09-08 10:56:05,499]: State: 2 - byte received (at 1631094965.4989734): 62
[DEBUG] [2021-09-08 10:56:05,499]: State: 3 - byte received (at 1631094965.4989734): 3
[DEBUG] [2021-09-08 10:56:05,499]: State: 4 - byte received (at 1631094965.4989734): 1
[DEBUG] [2021-09-08 10:56:05,499]: State: 7 - byte received (at 1631094965.4989734): 0
[DEBUG] [2021-09-08 10:56:05,500]: State: 8 - byte received (at 1631094965.4989734): 1
[DEBUG] [2021-09-08 10:56:05,500]: State: 8 - byte received (at 1631094965.4989734): 44
[DEBUG] [2021-09-08 10:56:05,500]: State: 9 - byte received (at 1631094965.4989734): 47
[DEBUG] [2021-09-08 10:56:05,500]: Message received (length 3) - Code 1
[DEBUG] [2021-09-08 10:56:05,500]: RAW message sent: bytearray(b'$M<\x00\x0
```2\x02')
[DEBUG] [2021-09-08 10:56:05,509]: State: 0 - byte received (at 1631094965.5091374): 36
[DEBUG] [2021-09-08 10:56:05,509]: State: 1 - byte received (at 1631094965.5091374): 77
[DEBUG] [2021-09-08 10:56:05,509]: State: 2 - byte received (at 1631094965.5091374): 62
[DEBUG] [2021-09-08 10:56:05,509]: State: 3 - byte received (at 1631094965.5091374): 4
[DEBUG] [2021-09-08 10:56:05,509]: State: 4 - byte received (at 1631094965.5091374): 2
[DEBUG] [2021-09-08 10:56:05,509]: State: 7 - byte received (at 1631094965.5091374): 66
[DEBUG] [2021-09-08 10:56:05,510]: State: 8 - byte received (at 1631094965.5091374): 84
[DEBUG] [2021-09-08 10:56:05,510]: State: 8 - byte received (at 1631094965.5091374): 70
[DEBUG] [2021-09-08 10:56:05,510]: State: 8 - byte received (at 1631094965.5091374): 76

... (quite big)

[DEBUG] [2021-09-08 10:56:05,595]: Message received (length 22) - Code 101
[DEBUG] [2021-09-08 10:56:05,595]: RAW message sent: bytearray(b'$M<\x00\x96\x96')
[DEBUG] [2021-09-08 10:56:05,600]: State: 0 - byte received (at 1631094965.6009123): 36
[DEBUG] [2021-09-08 10:56:05,601]: State: 1 - byte received (at 1631094965.6009123): 77
[DEBUG] [2021-09-08 10:56:05,601]: State: 2 - byte received (at 1631094965.6009123): 62
[DEBUG] [2021-09-08 10:56:05,601]: State: 3 - byte received (at 1631094965.6009123): 22
[DEBUG] [2021-09-08 10:56:05,601]: State: 4 - byte received (at 1631094965.6009123): 150
[DEBUG] [2021-09-08 10:56:05,603]: State: 7 - byte received (at 1631094965.6009123): 46
[DEBUG] [2021-09-08 10:56:05,603]: State: 8 - byte received (at 1631094965.6009123): 1
[DEBUG] [2021-09-08 10:56:05,603]: State: 8 - byte received (at 1631094965.6009123): 0
[DEBUG] [2021-09-08 10:56:05,603]: State: 8 - byte received (at 1631094965.6009123): 0
[DEBUG] [2021-09-08 10:56:05,603]: State: 8 - byte received (at 1631094965.6009123): 33
[DEBUG] [2021-09-08 10:56:05,603]: State: 8 - byte received (at 1631094965.6009123): 0
[DEBUG] [2021-09-08 10:56:05,603]: State: 8 - byte received (at 1631094965.6009123): 0
[DEBUG] [2021-09-08 10:56:05,603]: State: 8 - byte received (at 1631094965.6009123): 0
[DEBUG] [2021-09-08 10:56:05,604]: State: 8 - byte received (at 1631094965.6009123): 0
[DEBUG] [2021-09-08 10:56:05,604]: State: 8 - byte received (at 1631094965.6009123): 0
[DEBUG] [2021-09-08 10:56:05,604]: State: 8 - byte received (at 1631094965.6009123): 0
[DEBUG] [2021-09-08 10:56:05,604]: State: 8 - byte received (at 1631094965.6009123): 65
[DEBUG] [2021-09-08 10:56:05,604]: State: 8 - byte received (at 1631094965.6009123): 0
[DEBUG] [2021-09-08 10:56:05,604]: State: 8 - byte received (at 1631094965.6009123): 3
[DEBUG] [2021-09-08 10:56:05,604]: State: 8 - byte received (at 1631094965.6009123): 0
[DEBUG] [2021-09-08 10:56:05,604]: State: 8 - byte received (at 1631094965.6009123): 0
[DEBUG] [2021-09-08 10:56:05,605]: State: 8 - byte received (at 1631094965.6009123): 26
[DEBUG] [2021-09-08 10:56:05,605]: State: 8 - byte received (at 1631094965.6009123): 4
[DEBUG] [2021-09-08 10:56:05,605]: State: 8 - byte received (at 1631094965.6009123): 0
[DEBUG] [2021-09-08 10:56:05,605]: State: 8 - byte received (at 1631094965.6009123): 0
[DEBUG] [2021-09-08 10:56:05,605]: State: 8 - byte received (at 1631094965.6009123): 1
[DEBUG] [2021-09-08 10:56:05,605]: State: 8 - byte received (at 1631094965.6009123): 0
[DEBUG] [2021-09-08 10:56:05,605]: State: 9 - byte received (at 1631094965.6009123): 211
[DEBUG] [2021-09-08 10:56:05,605]: Message received (length 22) - Code 150

ricardodeazambuja, 2021-09-09:
Hi,

Are you sure this log is from read_altitude.py? If you check the code, the line with the assert comes before anything else:
https://github.com/thecognifly/YAMSPy/blob/337a4e87fb441c4d761f83734a5d9d46dde0f272/Examples/read_altitude.py#L45

The log file is concatenated by default, so you need to jump to the end to check for something that just went wrong.

Also, when that error occurs the log file should show this message:
https://github.com/thecognifly/YAMSPy/blob/337a4e87fb441c4d761f83734a5d9d46dde0f272/yamspy/__init__.py#L882

Finally, double check the serial port configuration in read_altitude.py.

Cheers,
Ricardo

miguelwon, 2021-09-09:
I just removed that assert condition and started to work...

Sorry, I know it's not related but now I'm running the `simpleUI.py`example that is working fine, but it is not arming at all. I think is because I get `armingDisableFlags: ['BOOTGRACE']`. Any idea what might be the problem?
Thanks.

ricardodeazambuja, 2021-09-09:
Thanks for spotting the bug! That condition (board == 1) was just crazy as it should return 1 when it is NOT working :)

This 'BOOTGRACE' seems to be related to the minimum time you need to wait before arming.
https://oscarliang.com/quad-arming-issue-fix/

miguelwon, 2021-09-09:
Thanks. Just executed `set pwr_on_arm_grace = 0` but keeps showing. :(

ricardodeazambuja, 2021-09-09:
Is there a reason for you to use betaflight? inav works better if you want a flying robot :)
Anyway, this is the only place checking for it in the current betaflight master branch:
https://github.com/betaflight/betaflight/blob/b228531b4548c44264b213c980e8728cb506b891/src/main/fc/core.c#L274

ricardodeazambuja, 2021-09-09:
Sorry for the silly question, but did you save after ```set pwr_on_arm_grace = 0```?

miguelwon, 2021-09-09:
No reason. I'm quite a noob in the drone field. Thanks for the tip! I'll try INAV, then. ;)

miguelwon, 2021-09-09:
Yes, I save it. ;)

ricardodeazambuja, 2021-09-09:
I'm curious about this bug because I've been away from betaflight for a while, but I would like to make sure the library still works when using betaflight.
Which version of betaflight are you using? Which FC are you using? 
If could be related to DSHOT. Check this line:
https://github.com/betaflight/betaflight/blob/b228531b4548c44264b213c980e8728cb506b891/src/main/fc/core.c#L279
And it seems to boil down to this:
https://github.com/betaflight/betaflight/blob/56b5ac57c72d4a2de092eb98d61e93e1547e0ce4/src/main/drivers/dshot_command.c#L157

miguelwon, 2021-09-09:
Ok, I just figured it out. It was a silly error from my side. I was setting wrongly the ESC protocol (for my case it must be PWM). Now everything is working fine!
Thanks. For the library as well! I have tried a few and YAMSPy is the only one that is working fine.

closed 2021-09-09 by ricardodeazambuja