thecognifly/YAMSPy #15: communication with INAV

DanchoDrone, 2021-11-18:
Hi,
I wanted to ask regarding the communication with INAV and raspberry pi (3b) model.
I've downloaded the library and connected the wired accordingly, verified that FC TX is connected to Rpi RX and vice versa.
However, I don't see a communication, the "curses ?UX" is not loaded.
I think it's a problem with how I configured the FC via INAV configurator.
Attached two snapshots. Could you please help here?
Thanks
Dancho
[image]
[image]

DanchoDrone, 2021-11-20:
@ricardodeazambuja 
Hi Ricardo, I saw that you created this library, could you please advise?
Thanks in advance :)
Dancho

DanchoDrone, 2021-12-14:
anyone here?

ricardodeazambuja, 2021-12-15:
Sorry for the delay, but I was moving between continents + covid... I hope you can figure the mess :)

Simple things to check: 
1. Are you sure you are connecting things to the correct UART in the FC? 
2. Are you sure the TXs are connected to the RXs? This may not be intuitive at first, but you connect TX => RX and RX => TX.
3. Check if you are using the correct name for the serial connection in the RPI. Try ```ls -lh /dev/serial*``` and test them.

If nothing helps, please, post your questions here and I will try my best to help. Cheers!

DanchoDrone, 2021-12-26:
@ricardodeazambuja Thanks for the detailed answer. I've figured it out and now it works great.
Sorry for my delay also (covid is also here :/)
Dancho

DanchoDrone, 2022-01-04:
@ricardodeazambuja - Of course :)
I've carried out the following steps:
1. Enable serial communication via "sudo raspi-config"
2. Disable S0 service by writing " sudo systemctl disable serial-getty@ttyS0.service"
3. Connect the RPI board to the Flight controller to specific UART (FC_TX<-> rpi_RX and FC_RX<->rpi_TX)
4. at INAV configurator changed two things:
a. Ports \ change the specific UART to MSP with baud rate 115200
b. Receiver \ Change "Receiver mode" to MSP
5. clone this project with serial address "/ttyAMA0"

and it worked :)

closed 2022-01-04 by ricardodeazambuja