thecognifly/YAMSPy #16: Joystick Event Values Problem

sukarian, 2022-02-05:
So I recently had to get a new SD card with Raspbian because my old one stopped working for some reason. I downloaded the most recent versions of the linux kernel and python-evdev, and when I run jstest the event values are between -32765 and 32765, or thereabouts, as opposed to 0-255. When I move the sticks, the event values change as expected. However, when I run the joystick-async.py script, nothing happens when I move the stick. I altered the code values to work with the new event values, but still nothing. I inserted a print statement into the code, but moving the sticks does nothing, for the event values, or the RC commands. What should I do? Should I just revert to an older version of the python-evdev package and the linux kernel? Or do I need to configure python-evdev somehow?

ricardodeazambuja, 2022-02-08:
I don't have a joystick right now, so I can't test if something changed. Check the output for your joystick using this:   
https://python-evdev.readthedocs.io/en/latest/tutorial.html#listing-device-capabilities-devices-with-absolute-axes
If you are getting higher resolution (16bits instead of 8bits), this is a good thing! You just need to adapt joystick-async.py to take that into account (e.g. add 32765 to make it start at zero again and use 2*32765 instead of 255). However, maybe the event.type values also changed, so you should check everything using the link above (and don't test this code with propellers mounted on a drone...)

sukarian, 2022-02-12:
I resolved the issue. It was an easy fix; I just had to check the device list, the name was updated from /dev/input/event2 -> /dev/input/event3. Thanks!

closed 2022-02-12 by sukarian