thecognifly/YAMSPy #17: Issues with arming using custom code

sukarian, 2022-02-27:
I've been trying to follow the provided examples to arm my quad using my custom code. Here's my code: 
------------------------------------------------------------------
#!!!
import time
from yamspy import MSPy

CMDS = {
            'roll':     1500,
            'pitch':    1500,
            'throttle': 900,
            'yaw':      1500,
            'aux1':     1000,
            'aux2':     1000
            #'aux3':     1000,
            #'aux4':     1000
            }

#CMDS = [1500,1500,900,1500,1000,1000]
CMDS_ORDER = ['roll', 'pitch', 'throttle', 'yaw', 'aux1', 'aux2']

serial_port = "/dev/ttyUSB0"
with MSPy(device=serial_port, loglevel='WARNING', baudrate=115200) as board:
    command_list = ['MSP_API_VERSION', 'MSP_FC_VARIANT', 'MSP_FC_VERSION', 'MSP_BUILD_INFO',
                    'MSP_BOARD_INFO', 'MSP_UID', 'MSP_ACC_TRIM', 'MSP_NAME', 'MSP_STATUS',
                    'MSP_STATUS_EX','MSP_BATTERY_CONFIG', 'MSP_BATTERY_STATE', 'MSP_BOXNAMES']
    for msg in command_list:
        if board.send_RAW_msg(MSPy.MSPCodes[msg], data=[]):
            dataHandler = board.receive_msg()
            board.process_recv_data(dataHandler)
    try:
        print('Arming')
        CMDS_RC = [CMDS[ki] for ki in CMDS_ORDER]
        #board.fast_msp_rc_cmd(CMDS_RC)
        while True:
            CMDS['aux1'] = 1800
            CMDS['throttle'] = 1000
            CMDS_RC = [CMDS[ki] for ki in CMDS_ORDER]
            if board.send_RAW_RC(CMDS_RC):
                dataHandler = board.receive_msg()
                board.process_recv_data(dataHandler)
            #board.fast_msp_rc_cmd(CMDS_RC)

            CMDS['throttle'] = 1050
            CMDS_RC = [CMDS[ki] for ki in CMDS_ORDER]
            board.fast_msp_rc_cmd(CMDS_RC)

    except KeyboardInterrupt:
        print('Disarming')
        CMDS['aux1'] = 900

        CMDS_RC = [CMDS[ki] for ki in CMDS_ORDER]
        if board.send_RAW_RC(CMDS_RC):
            dataHandler = board.receive_msg()
            board.process_recv_data(dataHandler)
        #board.fast_msp_rc_cmd(CMDS_RC)
-------------------------------------------------------------

My FC (an old Naze32) flashes green, indicating it's armed, but the motors won't spin. This isn't a problem for either the joystick_async.py or simpleUI.py code; both of those can arm and spin the motors. What needs to be done to arm the motors? I can't seem to find any obvious discrepancies between the example and my code.

ricardodeazambuja, 2022-03-01:
Hi. It would need more details to know why it's not arming. I suggest checking the blackbox (if your flight controller has one) and the configuration using the CLI.
Usually, the flight controller only arms after it had received some commands, so it will refuse to arm if the first command received is the one for arming. That could explain why the other examples are working. Please, let me know whether that was the problem.

ricardodeazambuja, 2022-04-29:
@sukarian may I close this issue?

closed 2022-10-17 by ricardodeazambuja