thecognifly/YAMSPy #22: Switch between RC and MSP drone contrling on flight

clashpunk, 2023-06-21:
I have succefully started simpleUI.py on my SpeedyBee f405 FC via USB cable. My question, is it possible to start manual flight using RC controller and then switch for autonomus controlling with Rasberry PI + YASMPy during flight?

ricardodeazambuja, 2023-06-25:
@clashpunk, I've been away from inav for a while, so I am not super aware of the bleeding edge, but I know it now has SITL and HITL, maybe they even have a new flight mode more like arducopter where you can enable autonomous control as you want.
What we have been doing is to control everything from the on-board computer (RPI or other). When we want manual control, we send the commands to the on-board computer and it sends the commands to the FC as if it was an MSP-based RC. Our safety is based on the failsafe mode because as soon as the on-board computer stops sending commands to the FC, the FC will enter into failsafe. 
You can also have a look at the [msp_override_channels](https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#msp_override_channels), it may be useful if you want to keep control of certain things (I haven't tried it yet).

devnim, 2024-01-14:
@ricardodeazambuja, I just want to clarify. If a Pi is connected to a betaflight FC, can a user still manually take off with his usual RC transmitter and fly the drone manually while a PI is connected and running this software on betaflight FC? If so, in this case, can the PI take over mid-flight (e.g. if some condition is met) and send its own RC commands to fly the drone autonomously? Thank you for your work on this.

ricardodeazambuja, 2024-01-14:
Have a look at this (old) discussion: https://github.com/betaflight/betaflight/issues/8292