thecognifly/YAMSPy #23: How to communicate with FC fast?

yrik, 2023-10-01:
When I'm running commands like `MSP_BOARD_INFO` it takes good 5-10 seconds to process each command. Is that normal?
What is the proper way to get fast accel/gyro values from FC? Say 100 times per second.

ricardodeazambuja, 2023-10-02:
My first suggestion is to try this branch: https://github.com/thecognifly/YAMSPy/tree/yb/stable
Then check the `Examples` folder.

ricardodeazambuja, 2024-05-22:
Hi @Tuxliri. The max frequency you can get depends on many factors. What is the microcontroller that your FC uses? Is it F7 or F4? You can also have a look at the frequencies reported by the inav-configurator to have an idea.
We use debug messages to speed things up (example from yb/stable branch: https://github.com/thecognifly/YAMSPy/blob/yb/stable/Examples/test_debug_pos_est.py). It's quite easy to modify inav to create new custom debug messages (https://github.com/thecognifly/inav/commit/8da90c9a4beb618ebe2ed798eb7e2db6e1f1c3af)
I prefer to use messages as high level as possible instead of directly reading the acc+gyro (I'm considering inav, not betaflight) because the FC has already done the heavy lifting for you (filtering...).

Tuxliri, 2024-05-29:
Thanks for the quick reply @ricardodeazambuja , I have been using Betaflight on an F4 board (SPEEDYBEEF405V3), did you manage to achieve high-frequency communication with it?

I am considering moving to iNAV or Arducopter (using MAVLink) if this is not possible.

ricardodeazambuja, 2024-05-30:
If you want to use MSP and cheap flight controllers, iNAV is the best option.