thecognifly/YAMSPy #26: The code works correctly or not on my microcontroller F411 NOXE V3

brightproject, 2024-02-29:
Thanks @ricardodeazambuja for the useful code.
I run the example code `simpleUI.py`, the result is like this
```
Press 'q' to quit, 'r' to reboot, 'm' to change mode, 'a' to arm, 'd' to disarm and arrow keys to control

ARMED: False                                      armingDisableFlags: ['RXLOSS', 'ANGLE', 'MSP']
cpuload: 31                                       cycleTime: 249
mode: 0                                           Flight Mode: []
Battery Voltage: 0.01V  VOLTAGE TOO HIGH
Motor Values: [1000, 1000, 1000, 1000, 0, 0, 0, 0]
RC Channels Values: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
GUI cycleTime: 18.98ms (average 74.16Hz)

apiVersion: 1.45.0                                flightControllerIdentifier: BTFL
flightControllerVersion: 4.4.3                    boardIdentifier: S411
boardName: JHEF411                                name:
```
[image]
If I select the `ARM/DISARM` command, it’s as if the command is not being sent, and I have questions in the `RX` input of my flight controller.
```
Sending Arm command...
ARMED: False
```
or
```
Sending Disarm command...
ARMED: False
```
Armed never changes, as if the command does not reach the pilot’s microcontroller.
But when I select the `REBOOT` command, the pilot reboots.
I don’t yet have enough knowledge in betaflight firmware, as far as I understand, the AWP flag appears in three cases:
_1. When there is no connection with the transmitter
2. When the accelerometer sensor does not show the horizon, it is not calibrated
3. There is no connection via the `MSP` protocol_.
I don't have a pult/control panel, ESC board or motors.
How is it possible to put into operation a “naked” flight controller without peripherals?

ricardodeazambuja, 2024-03-03:
Hi,

I'm sorry, but I haven't used Betaflight for a long time (I switched to iNAV). The reason you can't arm seems to be that you have some `armingDisableFlags` showing up.  `ANGLE` probably means your FC is not levelled (there's a max. angle that BF will accept to arm your drone, maybe 15degrees...). `RXLOSS` means you haven't set the RX properly, so BF is complaining it can't see the heartbeat from the RX. Finally, `MSP` probably means you can't arm your drone while using the USB port, unless you disable some safety flags (but, as the name suggests, this will make your drone unsafe). For a `naked` FC (no motors + propellers), you don't really care about the safety flags, therefore you can just disable them allowing the FC to arm from the USB.
In the CLI (unless they changed it in the newest versions), you should use `feature RX_MSP` to enable the RX as MSP.

Last time I used BF it was version 4.0.0, so they may have changed a lot of stuff compared to 4.4.3.

Here are some links that I hope they will help you:
https://betaflight.com/docs/wiki
https://betaflight.com/docs/wiki/guides/current/Arming-Sequence-And-Safety
https://betaflight.com/docs/wiki/guides/current/Runaway-Takeoff-Prevention
https://betaflight.com/docs/wiki/guides/current/Betaflight-4.0-CLI-commands
https://oscarliang.com/quad-arming-issue-fix/

brightproject, 2024-03-03:
Thanks @ricardodeazambuja for the links, especially for the `CLI` commands.
I understand more about the code, and then about the configurator and WIKI.
Unfortunately, I still can’t figure out how to start recording on a `black box`, which is written to the memory built into the board from a flight controller, without a control panel, `ESC` and motors.

ricardodeazambuja, 2024-03-03:
To record to the blackbox you may have two options: enable the FC to arm (after properly setting the blackbox using the betaflight-configurator) or by setting one "switch" that will start the recording (again, you need to use the configurator for that). BTW, the ARM command in the SimpleUI is expecting that you set the arm to the correct switch because the order of the values you send to the drone will take the switches into consideration.

brightproject, 2024-03-04:
That is, without a `physical` switch, is it difficult to `disarm` the flight controller?
Is it better to solder a toggle switch to the `FC` board and `disarm` it?

ricardodeazambuja, 2024-03-04:
It's not a physical switch, it's just software, but you need to enable the switch using betaflight-configurator. The RC command is typically AERT (roll, pitch, yaw, throttle) + the "switches" (called channel or aux).
https://oscarliang.com/betaflight-modes/
https://oscarliang.com/channel-map/